MotorolaMobilityLLC/kernel-msm #394: GPL | Source request for Motorola Moto Edge X30 (hiphic) S0SC32.16-49-4

Hasaber8, 2022-02-04:
Kindly requesting sources for the following repos for this tag "S0SC32.16-49-4"

kernel-msm
kernel-devicetree
kernel-camera-devicetree
kernel-display-devicetree
kernel-msm-5.10-techpack-camera
kernel-msm-5.10-techpack-display
kernel-msm-5.10-techpack-video
vendor-qcom-opensource-audio-kernel
vendor-qcom-opensource-wlan-qcacld-3.0
vendor-qcom-opensource-wlan-fw-api
vendor-qcom-opensource-wlan-qca-wifi-host-cmn
and with 5.10 all the components are split as far as I know, but not sure how they are split, please do upload everything.
Thanks in advance!
~Rohan

CC: @chenyt9 @awheeler-mmi

betaxab, 2022-03-25:
@mbissaromoto I need S1SC32.52-27-2 please.

chenyt9, 2022-03-25:
@betaxab @Hasaber8 ，
this version is already in our process and will be released as soon as possible. Under normal circumstances, itusually takes two weeks. Thank you!

chenyt9, 2022-05-06:
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S0SC32.16-49-4

closed 2022-05-06 by chenyt9