MotorolaMobilityLLC/kernel-msm #396: [REQUEST] Moto g20 (java) RTA31.68-29 sources

godbru, 2022-02-09:
Please take our request into consideration, a new kernel update has just come out if possible to source this thousands of users and devs will be grateful @awheeler-mmi @chenyt9 @mbissaromoto 🙏

godbru, 2022-02-10:
bump @awheeler-mmi @chenyt9 @mbissaromoto

l-gthb, 2022-02-19:
aren't they legally forced to provide sources?

SantiiAgoo5121, 2022-05-03:
@awheeler-mmi @chenyt9 @mbissaromoto 🙏🏻🙏🏻🙏🏻

Claravallac, 2023-02-21:
BUILD ID: RTAS31.68-66-2

please release all GPL protected linux kernel source code for Moto G20 device

@chenyt9
@awheeler-mmi
@TinnoOSS
@asbraga-motorola
@mbissaromoto

As npjohnson said:
"This is getting bordererline absurd. Especially considering this looks like EOL for this device family - please push it."

TinnoOSS, 2023-03-21:
@ALL,version released:
https://github.com/MotorolaMobilityLLC/kernel-sprd/releases/tag/MMI-RTA31.68-29