MotorolaMobilityLLC/kernel-msm #402: [Request] Motorola G 71 5G (corfur) Kernel Sources (RRUB31.Q3-71-102-2)

Venkat3620, 2022-04-15:
please publish kernel ,dts,techpack drivers for Motorola G 71 5G (corfur), @chenyt9 @awheeler-mmi @mbissaromoto

chenyt9, 2022-04-22:
Hi,@Venkat3620 , could you please inform us the version number ?

Venkat3620, 2022-04-22:
RRUB31.Q3-71-102-2

Venkat3620, 2022-04-26:
@chenyt9 , there is version of MMI-RRUB31.Q3-71-32-1R1 for corfur published yesterday ,is there any chance of pushing the latest one i.e, RRUB31.Q3-71-102-2

chenyt9, 2022-05-11:
@Venkat3620 ,sorry for miss the msg, will start to arrange the publish work for RRUB31.Q3-71-102-2

Venkat3620, 2022-05-11:
@chenyt9 thanks for response, previous tag has kernel-display-devicetree,kernel-camera-devicetree missing so please add them with this version.

Venkat3620, 2022-05-19:
@chenyt9 @awheeler-mmi @mbissaromoto please release the latest tag. RRUBS31.Q3-71-102-3

chenyt9, 2022-06-20:
@Venkat3620 ，
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/RRUB31.Q3-71-102-2
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/RRUBS31.Q3-71-102-2-3

closed 2022-06-20 by chenyt9