MotorolaMobilityLLC/kernel-msm #404: [Request] GPL | Motorola Moto E20 and E40 Source release (Device Tree, Kernel code and vendor, etc.) and unlockable bootloaders

iFUCKINGHATEcomputers, 2022-04-20:
A lot more life and use could be squeezed out of these models if they were made more open and customizable. 

Furthermore, the E20's bootloader doesn't seem to be unlockable at all - it merely gives errors such as `FAILED (remote: unknown cmd.)`. Others report the same error on the E40:
https://forums.lenovo.com/t5/MOTOROLA-Android-Developer-Community/Moto-E20-Bootloader-Unlock/m-p/5136560
https://forum.xda-developers.com/t/failed-remote-unknown-command-trying-to-unlock-bootloader.3841588/post-86474265

iFUCKINGHATEcomputers, 2022-04-20:
Related: #399

asbraga-motorola, 2022-08-04:
Hi. We do not generally support unlocking bootloaders. Hope that other users can help you in the developer community forum.

closed 2022-08-04 by asbraga-motorola

ThePedroo, 2025-02-02:
Hello. While the Moto E20 bootloader appears to be unlockable, like the author of this issue has stated, it would be beneficial to have its device tree, kernel source, vendor, among other components publicly available anyway, as required by GPL.

ThePedroo, 2025-02-03:
> Could you please open a new request with the exact build id you want?

Please follow in #673 issue.