MotorolaMobilityLLC/kernel-msm #405: [QUESTION] Help with MMI-RRF31.Q1-59-33 kernel compilation

decaduto, 2022-04-23:
Hello all,
I unlocked the bootloader of my moto g50 device, and I was trying to flash a fresh kernel, navigating on github, I found that the kernel I need is under the MMI-RRF31.Q1-59-33 tag, I am compiling the kernel with clang, precisely with the 6443078-11.0.1 version, which is the one present in the kernel version string, and I use as .config file, the one I found in the /proc/config.gz in my phone (I do not do "Make defconfig", instead I copied the .config in the main kernel folder and I start with make -j4 ARCH=arm64 CC=<clang path> CROSS_COMPILE=aarch64-linux-gnu-), now, even if the kernel compiles without any problem, when I try to flash in my phone (with fastboot flash boot <kernel>) it enters in bootloop, note that the size of my custom kernel differs of 9-10 mb in comparison with the stock kernel.

Am I missing something in the compilation phase? Or is the kernel source incomplete?

decaduto, 2022-04-25:
bump @vinimed

mbissaromoto, 2022-04-26:
Hi @decaduto 

We are investigating it here. Hope to have some answers soon...

decaduto, 2022-04-26:
Nice to know, I want to point out that the current stock kernel on my motorola g50 is the MMI-RRF31.Q1-59-33-2 version.

closed 2022-08-01 by asbraga-motorola