MotorolaMobilityLLC/kernel-msm #409: [Request] Motorola G 30 (caprip) Kernel Sources (S0RCS32.41-10-9-2)

Venkat3620, 2022-04-26:
please publish kernel ,dts,techpack drivers for Motorola G 30 (caprip) @chenyt9 @awheeler-mmi @mbissaromoto
tag- S0RCS32.41-10-9-2

decaduto, 2022-04-26:
Sorry one question: what is the techpack? in my kernel there isn't anything similar

decaduto, 2022-04-26:
so techpack is a "pack" of various specific drivers for qualcomm chipset? why my kernel compiles without any problems using the .config file found in /proc/config.gz, even if the techpack folder did not exists? (my kernel is 5.4 )

Venkat3620, 2022-06-20:
@chenyt9  @awheeler-mmi @mbissaromoto Sir any update?

asbraga-motorola, 2022-08-17:
@Venkat3620 
Version is released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S0RCS32.41-10-9-2

closed 2022-08-17 by asbraga-motorola