MotorolaMobilityLLC/kernel-msm #414: missing headers in msm-kernel-modules-MMI-RRF31.Q1-59-33

decaduto, 2022-05-09:
Hello again,

I have downloaded the msm-kernel-modules-MMI-RRF31.Q1-59-33 package and then extracted in the main kernel directory (MMI-RRF31.Q1-59-33), then I added to the Makefile under drivers/Makefile the following lines:

```
obj-y                           += bus/ sensors/ backlight/ktd3136/

obj-$(CONFIG_GENERIC_PHY)       += phy/ uwb/sr100/  gud/MobiCoreDriver/ fm/
```

Just for see if the compilation failed or not.

Here some of the numerous errors I encountered:

```
drivers/gud/MobiCoreDriver/platform.h:33:10: fatal error: 'soc/qcom/qtee_shmbridge.h' file not found
#include <soc/qcom/qtee_shmbridge.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [scripts/Makefile.build:292: drivers/gud/MobiCoreDriver/clock.o] Errore 1
make[2]: *** Attesa per i processi non terminati....
In file included from drivers/gud/MobiCoreDriver/fastcall.c:26:
drivers/gud/MobiCoreDriver/platform.h:33:10: fatal error: 'soc/qcom/qtee_shmbridge.h' file not found
#include <soc/qcom/qtee_shmbridge.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

```
drivers/gud/MobiCoreDriver/platform.h:30:10: fatal error: 'soc/qcom/scm.h' file not found
#include <soc/qcom/scm.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [scripts/Makefile.build:292: drivers/gud/MobiCoreDriver/clock.o] Errore 1
make[2]: *** Attesa per i processi non terminati....
In file included from drivers/gud/MobiCoreDriver/fastcall.c:24:
drivers/gud/MobiCoreDriver/mci/mcifc.h:18:10: fatal error: 'platform.h' file not found
#include "platform.h"
         ^~~~~~~~~~~~
1 error generated.
make[2]: *** [scripts/Makefile.build:292: drivers/gud/MobiCoreDriver/fastcall.o] Errore 1
In file included from drivers/gud/MobiCoreDriver/iwp.c:40:
drivers/gud/MobiCoreDriver/mci/mcimcp.h:19:10: fatal error: 'mci/mcloadformat.h' file not found
#include "mci/mcloadformat.h"
         ^~~~~~~~~~~~~~~~~~~~
1 error generated.
```

is clear that some headers are missing, this is only one of the few errors I get (others where related to the missing fw_name/extended_report field in ```syna_tcm_board_data``` in _include/linux/input/synaptics_tcm.h_ )

I also get the following dtb warnings when I compile the kernel:

```
[emantovani@111111111111 ibiza-stock]$ sudo make CC=/home/emantovani/Scaricati/android_prebuilts_clang_host_linux-x86_clang-6443078-11.0.1/bin/clang CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 -j4

  DTC     arch/arm64/boot/dts/vendor/qcom/holi-moto-base.dtb
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1175.5-15: Warning (reg_format): /soc/funnel@8863000/out-ports/port@0:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1184.5-15: Warning (reg_format): /soc/funnel@8863000/out-ports/port@1:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1193.5-15: Warning (reg_format): /soc/funnel@8863000/out-ports/port@2:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1460.5-15: Warning (reg_format): /soc/funnel@8a5f000/out-ports/port@0:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1469.5-15: Warning (reg_format): /soc/funnel@8a5f000/out-ports/port@1:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:2012.5-15: Warning (reg_format): /soc/replicator@8046000/out-ports/port@0:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/vendor/qcom/holi-moto-base.dtb: Warning (pci_device_bus_num): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/vendor/qcom/holi-moto-base.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/vendor/qcom/holi-moto-base.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1174.11-1181.6: Warning (avoid_default_addr_size): /soc/funnel@8863000/out-ports/port@0: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1174.11-1181.6: Warning (avoid_default_addr_size): /soc/funnel@8863000/out-ports/port@0: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1183.11-1190.6: Warning (avoid_default_addr_size): /soc/funnel@8863000/out-ports/port@1: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1183.11-1190.6: Warning (avoid_default_addr_size): /soc/funnel@8863000/out-ports/port@1: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/holi-coresight.dtsi:1192.11-1199.6: Warning (avoid_default_addr_size): /soc/funnel@8863000/out-ports/port@2: Relying on default #address-cells value
[... 205 lines cut ...]
arch/arm64/boot/dts/vendor/qcom/pm6150l.dtsi:394.21-401.6: Warning (avoid_default_addr_size): /fragment@4/__overlay__/qcom,pm6150l@5/qcom,amoled/ab@de00: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/pm6150l.dtsi:403.23-410.6: Warning (avoid_default_addr_size): /fragment@4/__overlay__/qcom,pm6150l@5/qcom,amoled/ibb@dc00: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/pm6150l.dtsi:403.23-410.6: Warning (avoid_default_addr_size): /fragment@4/__overlay__/qcom,pm6150l@5/qcom,amoled/ibb@dc00: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/holi-pmic-overlay.dtsi:174.35-194.5: Warning (avoid_default_addr_size): /fragment@7/__overlay__/qcom,pm6150l@4/adc_tm@3400: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/holi-pmic-overlay.dtsi:174.35-194.5: Warning (avoid_default_addr_size): /fragment@7/__overlay__/qcom,pm6150l@4/adc_tm@3400: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:36.14-44.4: Warning (avoid_default_addr_size): /fragment@31/__overlay__/ramoops_mem: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:36.14-44.4: Warning (avoid_default_addr_size): /fragment@31/__overlay__/ramoops_mem: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:46.38-50.4: Warning (avoid_default_addr_size): /fragment@31/__overlay__/wdog_cpuctx_region: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:46.38-50.4: Warning (avoid_default_addr_size): /fragment@31/__overlay__/wdog_cpuctx_region: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:52.40-56.4: Warning (avoid_default_addr_size): /fragment@31/__overlay__/tzlog_dump_mem_region: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:52.40-56.4: Warning (avoid_default_addr_size): /fragment@31/__overlay__/tzlog_dump_mem_region: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:58.44-62.4: Warning (avoid_default_addr_size): /fragment@31/__overlay__/mmi_annotate_mem_region: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:58.44-62.4: Warning (avoid_default_addr_size): /fragment@31/__overlay__/mmi_annotate_mem_region: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:207.19-234.4: Warning (avoid_default_addr_size): /fragment@39/__overlay__/synaptics_tcm@20: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:207.19-234.4: Warning (avoid_default_addr_size): /fragment@39/__overlay__/synaptics_tcm@20: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:236.19-339.4: Warning (avoid_default_addr_size): /fragment@39/__overlay__/sx937x@2c: Relying on default #address-cells value
arch/arm64/boot/dts/vendor/qcom/holi-moto-common-overlay.dtsi:236.19-339.4: Warning (avoid_default_addr_size): /fragment@39/__overlay__/sx937x@2c: Relying on default #size-cells value
arch/arm64/boot/dts/vendor/qcom/ibiza-touchscreen.dtsi:76.11-88.4: Warning (avoid_default_addr_size): /fragment@50/__overlay__/novatek@0: Relying on default #address-cells value
ECC... (it continues, but Github has a limit of words, unfortunately)

```

(This after having merged the folder 'boot/dts' with the various kernel-devicetree repos)

Please, could you upload the **FULL** kernel sources, instead of some pieces with missing parts?

mbissaromoto, 2022-05-10:
@decaduto 
What is the kernel version of this machine where you're trying to build it? 
Using the first error you reported:

```
drivers/gud/MobiCoreDriver/platform.h:33:10: fatal error: 'soc/qcom/qtee_shmbridge.h' file not found
#include <soc/qcom/qtee_shmbridge.h>
```

This platform.h file has it on top:

```
#if KERNEL_VERSION(4, 19, 0) <= LINUX_VERSION_CODE
#define USE_SHM_BRIDGE
#endif

#include <soc/qcom/scm.h>
#include <soc/qcom/qseecomi.h>
#if defined USE_SHM_BRIDGE
#include <soc/qcom/qtee_shmbridge.h>
#endif
```

So, it is trying to get this soc/qcom/qtee_shmbridge.h (because USE_SHM_BRIDGE is defined as true). This folder does not exists on kernel 5.4. For kernel 5.4 it is here https://github.com/MotorolaMobilityLLC/kernel-msm/blob/MMI-RRF31.Q1-59-33/include/linux/qtee_shmbridge.h

decaduto, 2022-05-24:
Hello, I cannot find what is the correct watchdog driver in drivers/ folder there is watchdogtest and watchdog_cpu_ctx, the first is compiled without any errors, while the seconds return the following issue during the compilation:
```
  CC [M]  drivers/watchdog_cpu_ctx/watchdog_cpu_ctx.o
drivers/watchdog_cpu_ctx/watchdog_cpu_ctx.c:284:18: error: use of undeclared identifier 'VA_START'
        return (addr >= VA_START);
                        ^
drivers/watchdog_cpu_ctx/watchdog_cpu_ctx.c:822:14: error: implicit declaration of function 'dma_remap' [-Werror,-Wimplicit-function-declaration]
        ctx_vaddr = dma_remap(dev, NULL, pdata->mem_address,
                    ^
drivers/watchdog_cpu_ctx/watchdog_cpu_ctx.c:822:14: note: did you mean 'memremap'?
./include/linux/io.h:156:7: note: 'memremap' declared here
void *memremap(resource_size_t offset, size_t size, unsigned long flags);
      ^
drivers/watchdog_cpu_ctx/watchdog_cpu_ctx.c:822:12: warning: incompatible integer to pointer conversion assigning to 'struct msm_wdog_cpuctx *' from 'int' [-Wint-conversion]
        ctx_vaddr = dma_remap(dev, NULL, pdata->mem_address,
                  ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```c

decaduto, 2022-07-01:
up @mbissaromoto

decaduto, 2022-07-01:
up @mbissaromoto

chenyt9, 2022-08-03:
Hi,@decaduto , No code is missing. Pls build refer to this readme file:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-RRF31.Q1-59-33.txt

closed 2022-08-03 by chenyt9