MotorolaMobilityLLC/kernel-msm #418: MSM8953 devices to 4.9

Ratatouilledummie, 2022-05-27:
is possible to msm8953 devices be ported to 4.9 releases ?

chenyt9, 2022-07-27:
Hi,@Ratatouilledummie ，
Are you asking for open source kernel code? If yes，could you please describe in detail what’s product you need and the corresponding version number. Thanks.

FPSensor, 2022-10-21:
nno since Motorola only offers the kernel of the stock ROM and it is not convenient for them to port one but even so the community is lately porting them by hand, some examples are: Montana, Hannah, cedric (brokensensors) and Albus (charger broken)

asbraga-motorola, 2023-06-14:
I'm closing this issue because it does not request the code of a specific product.
For generic questions, you may try the developer community:
https://forums.lenovo.com/t5/MOTOROLA-Android-Developer-Community/bd-p/230

closed 2023-06-14 by asbraga-motorola