MotorolaMobilityLLC/kernel-msm #420: [Request] Release Moto G Stylus 5G 2022 XT2215-4 (S1SD32.56-22) Sources, Kernel, Device Tree, etc.

bripage, 2022-06-05:
Can you please release a tag for the 2022 (XT2215-4) model of the Moto G Stylus 5G containing the kernel, device tree, etc. 

I believe the software version to be S1SD32.56-22

Thank you very much!!!

GhettiGuru, 2022-07-20:
Requesting GPL sources/Device Trees for the XT2215-4 Motorola Moto G Stylus 2022 3rd generation codename MilanF., Build tag S1SD32.56-22. Thanks in advance!

PizzaG, 2022-07-23:
I am also requesting this so we can get dev progress moving on this awesome device!!

chenyt9, 2022-07-27:
@PizzaG @bripage @GhettiGuru 
The version you requested is in the process. Please be patient. Thanks.

PizzaG, 2022-07-27:
> @PizzaG @bripage @GhettiGuru The version you requested is in the process. Please be patient. Thanks.

Will do, thanks for the heads update

bripage, 2022-08-22:
Any update on this? Been a few months now...

PizzaG, 2022-08-22:
> Any update on this? Been a few months now...

BUMP

Software Version S1SD32.56.81

chenyt9, 2022-08-26:
Hi，@all
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SD32.56-81R1
Thank you！

PizzaG, 2022-08-26:
> Hi，@ALL Version released： https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SD32.56-81R1 And will release S1SD32.56-22 in the next week. Thank you！

Thank You Very Much, it's greatly appreciated!!  This is why I go with Motorola products and tell my friends about them, They still seem to realize what Android and the ecosystem is all about.  Having fun and learning!!

GhettiGuru, 2022-09-05:
> Hi，@ALL Version released： https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SD32.56-81R1 And will release S1SD32.56-22 in the next week. Thank you！

Very much appreciated, however could you provide a device link tree for running scripts with studio? Kernel source tree, super, 
(system, product, vendor) ect trees? That would make compiling a heck of a lot easier thank you so much for all your hard work you guys are awesome this is my like 5th or 6th Motorola device I've been with you guys since the very first Motorola Droid and I still think my Droid x was way ahead of the game built-in HDMI.. apps tailor made just for it lol.. every time I factory reset I get droid-x bootstrapper and various tools purchased from Google installed still to this day! Thanks guys

chenyt9, 2022-09-20:
S1SD32.56-22 version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SD32.56-22R3

closed 2022-09-20 by chenyt9

whitedavidp, 2024-07-24:
> > Hi，@ALL Version released： https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SD32.56-81R1 And will release S1SD32.56-22 in the next week. Thank you！

Can anyone tell me which, if either, of these is correct for build # S1sd32.56-81-16? I am thinking of taking a whack at building TWRP for my XT2215-4. I really miss having it. Thanks!

W0NLB, 2025-01-15:
hey there! im a bit late to the party, but will the above release work on the T2SDS33.75-38-16-6?  i have the stock files from Moto if needed.  Im a bit intimidated to atempt it on my own.  every TWRP guide im finding says MTK device on it, but im pretty sure this thing is a SM6375 so a bit worried it will not work, or worse i make a paperweight,  any help or guidance is appreciated, id love to do it my self, i guess im just unconfident where to start.  Thank you!