MotorolaMobilityLLC/kernel-msm #422: (GPL)(REQUEST) Kernel Source Release for RRYA31.Q3-56-99-2(Moto G51)

18712886438, 2022-06-14:
hi, can u please release the kernel source for moto g51? thx :p

chenyt9, 2022-07-27:
Hi, @18712886438 
The kernel source for moto g51 has been released：https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRYA31.Q3-56-79-3
 And the version you requested are also in the process.Please be patient. Thanks.

chenyt9, 2022-09-22:
Hi, @18712886438 ，
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRYA31.Q3-56-99-2
Thank you for your patience！

closed 2022-09-22 by chenyt9

18712886438, 2022-09-22:
> Hi, @18712886438 ，
> Version released：
> https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRYA31.Q3-56-99-2
> Thank you for your patience！

thx.