MotorolaMobilityLLC/kernel-msm #429: GPL request for sources for Moto G Power(2022)

Rondeau79, 2022-07-07:
Requesting GPL sources for Motorola G Power (2022) Tonga build tag RRQ531.Q3-68-140-2 
Kindly requesting sources for the following repos for this device.

kernel-msm
kernel-devicetree
kernel-camera-devicetree
kernel-display-devicetree
kernel-msm-4.19-techpack-camera
kernel-msm-4.19-techpack-display
kernel-msm-4.19-techpack-video
vendor-qcom-opensource-audio-kernel
vendor-qcom-opensource-wlan-qcacld-3.0
vendor-qcom-opensource-wlan-fw-api
vendor-qcom-opensource-wlan-qca-wifi-host-cmn
Thanks in advance!
Rondeau79

chenyt9, 2022-07-27:
@Rondeau79 
Tonga version already completed release：
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-RRQ31.Q3-68-16-2R1
The repos you listed are not Tonga's. Tonga is an MTK platform, not Qualcomm.

chenyt9, 2022-10-10:
Hi, @Rondeau79 ,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-RRQS31.Q3-68-140-2
Thank you for your patience.

closed 2022-10-10 by chenyt9