MotorolaMobilityLLC/kernel-msm #434: [Request] Moto G82 (rhodep_g XT2225-1) - S1SUS32.73-13-4-3 kernel/dts/techpack

sevenrock, 2022-08-05:
Kindly requesting source for the following repos for this tag "S1SUS32.73-13-4-3"

kernel-msm-5.4-techpack-audio
kernel-msm-5.4-techpack-camera
kernel-msm-5.4-techpack-display
kernel-msm-5.4-techpack-video

The rest of the kernel sources/modules/devicetrees/wlan were released for rhodep-cn
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SU32.73-34-7

Are they compatible with rhodep_g (XT2225-1)? If not, please release the rhodep_g sources as well.

Hasaber8, 2022-08-06:
CC: @chenyt9

sevenrock, 2022-08-06:
Never mind, I found the techpacks for rhodep-cn, unexpectedly, in the non-5.4 repos
https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-audio/releases/tag/MMI-S1SU32.73-34-7
https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-camera/releases/tag/MMI-S1SU32.73-34-7
https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-display/releases/tag/MMI-S1SU32.73-34-7
https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-video/releases/tag/MMI-S1SU32.73-34-7

But the question remains: Can they be used to rhodep_g (XT2225-1) as well?

closed 2022-08-06 by sevenrock