MotorolaMobilityLLC/kernel-msm #435: [REQUEST] Moto G82 5G S1SUS32.73-13-4-3 Sources, Kernel, Device Tree, etc.

XylenC4, 2022-08-06:
Hello, could you please release the Sources, Kernel, Device Tree, etc. for the Moto G82 5G from S1SUS32.73-13-4-3?

chenyt9, 2022-08-16:
@XylenC4 ，
The publishing process has been triggered, please be patient. Thank you!

adammover, 2022-09-10:
Please release the kernel sources for moto g82 5g

adammover, 2022-09-10:
> @XylenC4 ， The publishing process has been triggered, please be patient. Thank you!

There are 25 days since publishing process has started

chenyt9, 2022-09-20:
@XylenC4  @adammover ,
Pls use this version first,
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SU32.73-41-2-4
We are preparing to release another version. Thanks.

adammover, 2022-09-22:
> 

What about the trees?

chenyt9, 2022-09-26:
Hi, @all,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/S1SUS32.73-13-4-3

@adammover , devicetree is in other repos
kernel-devicetree
kernel-*-devicetree

Thank you all.

closed 2022-09-26 by chenyt9