MotorolaMobilityLLC/kernel-msm #436: Requesting for the 2nd time.

MelonMusk93, 2022-08-06:
Request to please release Kernel tree, vendor tree , device sources for Moto E40 
There's already an issue raised it seems no one's interested or even have the courtesy to reply. Goodjob moto support team. There are multiple posts regarding Unisoc Chipset device sources release but no one bothers to look into user requests. 

https://github.com/MotorolaMobilityLLC/kernel-mtk/issues/67#issue-1316181056

FuccDucc, 2023-04-16:
@chenyt9 @awheeler-mmi @mbissaromoto @asbraga-motorola @TinnoOSS

Motorola is violating the GPL laws

asbraga-motorola, 2023-09-13:
The code is published in repo kernel-sprd: https://github.com/MotorolaMobilityLLC/kernel-sprd/releases/tag/MMI-ROQS31.166-42-1

closed 2023-09-13 by asbraga-motorola