MotorolaMobilityLLC/kernel-msm #439: [REQUEST] Moto G62 (India Snapdragon 695) (S2SSI32.18-21-1) Sources, Kernel, Device Tree, etc

pudup, 2022-08-30:
Exactly as title suggests. Sorry if it's already out somewhere but I couldn't find it.

Forenche, 2022-09-08:
ETA rhodei kernel source.

pudup, 2022-09-23:
> ETA rhodei kernel source.

Here try this
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SU32.73-41-2-4

chenyt9, 2022-10-10:
@pudup @Forenche 
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S2SSI32.18-12-1
Thank you all for your patience!

chenyt9, 2022-10-10:
@pudup , 
I guess your version number is wrong, it should be S2SSI32.18-12-1, not S2SSI32.18-21-1.  Pls help double confirm. Thanks.

pudup, 2022-10-10:
@chenyt9 yeah I typed it wrong, it's 12-1. Thanks

closed 2022-10-10 by pudup

Forenche, 2022-10-10:
> @pudup @Forenche Version released： https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S2SSI32.18-12-1 Thank you all for your patience!

Thank you for your reply :D I was trying to to get the camera and USB working but then you guys released the updated source, thank you.

Forenche, 2022-10-11:
@chenyt9 HI, the tag you mentioned here apparently has a stack corruption at early init. I can give you the pstore if you want to go through but it's corrupted. I tried building the previous tag,  MMI-S1SU32.73-41-2-4,  with the modules, techpack and qcacld stack from the S2SSI32.18-12-1 tag and it boots, so the issue I guess is with the base itself. Both the kernels were compiled using the same toolchain, AOSP Clang(r468909b). Please do look into it and hopefully the issue is solved ASAP, thank you.