MotorolaMobilityLLC/kernel-msm #442: [Request] Please release kernel source for Moto Edge 30 Fusion

switch-gott, 2022-09-10:
Dir sir or madam,
Please release kernel source code for new device moto edge 30 fusion for android 12.
Realy thanks
Kind regards

switch-gott, 2022-09-14:
Please also release device tree etc. 
It is a little bit disappointing,  why it takes so much time to do this.

chenyt9, 2022-09-20:
Hi，@switch-gott ,
I need more information to make sure the released version meets your needs, so could you please provide me with which version you need? I will prioritize the release of this version, thanks!

switch-gott, 2022-09-20:
Hi,
many thanks for your response.
I have not yet received the device.
But it's about the europe version of the fusion, especially for germany.
But I think it should be: S3SJ32.1-44-6 (XT2243-1)
In any case, please release the latest version for the German device.

switch-gott, 2022-09-22:
Hi Moto,
Now please release the newest version:
S3SJ32.1-44-7

switch-gott, 2022-09-29:
Hello Moto,
what's going on there? 
Why does it take so long to provide the kernel sources?
If you want the android community, then you would have to reach out to them more as well.
It's a bit disappointing, because you build great smartphones, but
as far as the sources and the release cycles are concerned, there really is still some room for improvement :-(

chenyt9, 2022-10-10:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SJ32.1-44-6
We will trigger the release process for version S3SJ32.1-44-7. Each version may need to take about 2-3 weeks because of the approval process and the release confirmation. please wait patiently!

chenyt9, 2022-12-20:
@switch-gott ,S3SJ32.1-44-7 is here:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SJ32.1-44-7

chenyt9, 2022-12-21:
@switch-gott ,We will trigger the release process for S3SJ32.1-71.
Before release, you can use this version first: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SJ32.1-86-1

chenyt9, 2023-03-28:
@switch-gott 
For S3SJ32.1-71 pls refer to here
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SJ32.1-71
Thanks

Eduardocastrox, 2023-07-27:
Manda kernel do motoro Edge 30 fusion Android 12 versão por favor S3SJS32.1-86-1-4

asbraga-motorola, 2023-07-28:
@Eduardocastrox 
Could you please open a new issue requesting that version?