MotorolaMobilityLLC/kernel-msm #444: Req: guamp android 11 Kernel

FPSensor, 2022-09-14:
for research reasons I require the android 11 kernel of Moto G9 Play guamp, I await your response.
Greetings.

asbraga-motorola, 2022-09-15:
Hi @FPSensor,
Can you inform the exact released version you want?
You find it in the System Config in the phone. It is a group of letters and numbers, that look similar to these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2, ...

FPSensor, 2022-09-15:
I need RPXS31.Q2-58-17-4-8

FPSensor, 2022-10-21:
Well I made the request more than a month ago, I think it would be good to start knowing at least if they are working on it

Joxquin, 2022-10-23:
Please releasw the kernel source of guamp RPXS31.Q2-58-17-4-8

Fercho289, 2022-10-23:
Please releasw the kernel source of guamp RPXS31.Q2-58-17-4-8

Emersoou78, 2022-10-23:
Please releasw the kernel source of guamp RPXS31.Q2-58-17-4-8

kevios12, 2022-10-23:
Please releasw the kernel source of guamp RPXS31.Q2-58-17-4-8

FranciscoAlefe, 2022-10-23:
Up

pesejo69, 2022-10-23:
We need  RPXS31.Q2-58-17-4-8 please ;)

pedroborraz, 2022-10-23:
Please realese the kernel source for guamp

roynatech2544, 2022-10-24:
You should release kernel sources like other OEMs do.

asbraga-motorola, 2022-10-27:
We are working on it. I requested priority for this version.

asbraga-motorola, 2022-11-01:
Version is released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RPXS31.Q2-58-17-4-8

closed 2022-11-01 by asbraga-motorola