MotorolaMobilityLLC/kernel-msm #446: device tree release

Rogerjr442200, 2022-09-19:
if i could get a release of the actual device device tree that would be great .. please and thank you

asbraga-motorola, 2022-09-19:
Hi @Rogerjr442200,
Can you inform the exact released version you want?
You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2.

Rogerjr442200, 2022-09-19:
RZBS31.Q2-143-27-13 is that what you need?@https://github.com/asbraga-motorola

Rogerjr442200, 2022-09-20:
?? @asbraga-motorola

asbraga-motorola, 2022-09-20:
Yes, that's the version identification I need. Thanks.

asbraga-motorola, 2022-10-17:
Duplicated: https://github.com/MotorolaMobilityLLC/kernel-msm/issues/438.
Code is released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RZBS31.Q2-143-27-13

closed 2022-10-17 by asbraga-motorola