MotorolaMobilityLLC/kernel-msm #448: XT2215-4 MotoG Stylus 5g 2022 Kernel, system, vendor sources

GhettiGuru, 2022-09-20:
I know you're probably working on it but we need kernel and all other gpl device trees for build MMI-S1SD32.56-81R2 ota. As well as the deconfig file please.

chenyt9, 2022-12-21:
@GhettiGuru , I don't find this version S1SD32.56-81R2 , is the correct version number S1SD32.56-81R1?
 If yes, here is the version release ：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SD32.56-81R1

closed 2022-12-21 by chenyt9

GhettiGuru, 2022-12-21:
> @GhettiGuru , I don't find this version S1SD32.56-81R2 , is the correct version number S1SD32.56-81R1? If yes, here is the version release ： https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SD32.56-81R1

That's the initial release, hell now I need  56-81-5 ota kernel source..they keep pushing ots

Kernel version 5.4.147-moto-gfbc163a5fe93

#1 Thu Nov 3 01:33:34 CDT 2022

Build number

S1SDS32.56-81-5

> @GhettiGuru , I don't find this version S1SD32.56-81R2 , is the correct version number S1SD32.56-81R1? If yes, here is the version release ： https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SD32.56-81R1

Well I needed the 2nd ota, but now I need 56-81-5 kernel source. They keep pushing otas lol. So, NOW,  I need kernel source and deconfig for:

Kernel version 5.4.147-moto-gfbc163a5fe93

#1 Thu Nov 3 01:33:34 CDT 2022

Build number

S1SDS32.56-81-5