MotorolaMobilityLLC/kernel-msm #450: [Request] GPL - Motorola g31[retin] Source release - Device Tree, Kernel Tree and Vendor Tree/Blob

VTVISHWANATH, 2022-10-14:
[Request] GPL - Motorola g31[retin] Source release - Device Tree, Kernel Tree and Vendor Tree/Blob
For the latest build number: RRWBS31.Q3-46-85-5

VTVISHWANATH, 2022-10-25:
Do we have any update in Here @motorolamobility?

How many days it will take to release the source for Moto G31 phone?

chenyt9, 2022-12-21:
@VTVISHWANATH ，
Sorry, I missed this message, this version is already on our release list, thank you very much for your patience. We will publish it as soon as possible.
Here is a release for Motorola G31：
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-RRWB31.Q3-46-64-2-2
Can you please use this version first？

chenyt9, 2023-01-19:
Version released：
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-RRWBS31.Q3-46-85-7

closed 2023-01-19 by chenyt9