MotorolaMobilityLLC/kernel-msm #451: GPL | Source request for Motorola Edge 30 Ultra (eqs) (S3SQ32.16-53-9)

SGCMarkus, 2022-10-14:
Kindly requesting source for the following repos for this tag "S3SQ32.16-53-9"

kernel-msm
kernel-devicetree
kernel-audio-devicetree
kernel-camera-devicetree
kernel-cvp-devicetree
kernel-display-devicetree
kernel-mmrm-devicetree
kernel-video-devicetree
kernel-msm-5.10-techpack-camera
kernel-msm-5.10-techpack-cvp
kernel-msm-5.10-techpack-display
kernel-msm-5.10-techpack-eva
kernel-msm-5.10-techpack-mmrm
kernel-msm-5.10-techpack-sched
kernel-msm-5.10-techpack-video
vendor-qcom-opensource-audio-kernel
vendor-qcom-opensource-dataipa
vendor-qcom-opensource-datarmnet
vendor-qcom-opensource-datarmnet-ext
vendor-qcom-opensource-wlan-qcacld-3.0
vendor-qcom-opensource-wlan-fw-api
vendor-qcom-opensource-wlan-qca-wifi-host-cmn
motorola-kernel-modules

and others if applicable.

Thanks in advance!
Markus

mukul2259, 2022-10-14:
> Kindly requesting source for the following repos for this tag "S3SQ32.16-53-9"
>
> kernel-msm kernel-devicetree kernel-audio-devicetree kernel-camera-devicetree kernel-cvp-devicetree [...]

Bump

Pierre2324, 2022-10-17:
> Kindly requesting source for the following repos for this tag "S3SQ32.16-53-9"
>
> kernel-msm
> kernel-devicetree
> kernel-audio-devicetree
> kernel-camera-devicetree
> kernel-cvp-devicetree [...]

+1

SoulesSnow, 2022-10-29:
> Kindly requesting source for the following repos for this tag "S3SQ32.16-53-9"
>
> kernel-msm kernel-devicetree kernel-audio-devicetree kernel-camera-devicetree kernel-cvp-devicetree [...]

+1

SGCMarkus, 2022-10-31:
@chenyt9 @asbraga-motorola any updates on this?

ThEMarD, 2022-11-22:
I would also like source code release for eqs

mukul2259, 2022-12-16:
Any update on this?

chenyt9, 2022-12-21:
Hi, @all ,
We will trigger the release of this version S3SQ32.16-53-9.
could you use this version first:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SL32.16-72-11

chenyt9, 2022-12-21:
@all,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SQ32.16-53-9

closed 2022-12-21 by chenyt9

SGCMarkus, 2022-12-21:
> @ALL, Version released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SQ32.16-53-9

Thank you, but there are still some missing repositories that need to be updated.

kernel-audio-devicetree
kernel-cvp-devicetree
kernel-mmrm-devicetree
kernel-video-devicetree
kernel-msm-5.10-techpack-camera (alternatively kernel-msm-techpack-camera or vendor-qcom-opensource-camera-kernel)
kernel-msm-5.10-techpack-cvp (alternatively kernel-msm-techpack-cvp or vendor-qcom-opensource-cvp-kernel)
kernel-msm-5.10-techpack-display (alternatively kernel-msm-techpack-display or vendor-qcom-opensource-display-drivers)
kernel-msm-5.10-techpack-eva (alternatively kernel-msm-techpack-evaor vendor-qcom-opensource-eva-kernel)
kernel-msm-5.10-techpack-mmrm (alternatively kernel-msm-techpack-mmrm or vendor-qcom-opensource-mmrm-driver)
kernel-msm-5.10-techpack-sched (alternatively kernel-msm-techpack-sched or vendor-qcom-opensource-sched)
kernel-msm-5.10-techpack-video (alternatively kernel-msm-techpack-video or vendor-qcom-opensource-video-driver)
kernel-msm-5.10-techpack-dataipa (alternatively kernel-msm-techpack-dataipa or vendor-qcom-opensource-dataipa)
(alternate names based on previous projects here or on their naming on https://git.codelinaro.org/clo/la/platform/vendor/opensource)