MotorolaMobilityLLC/kernel-msm #452: GPL | Source request for Motorola Moto Edge (XT2143-1) RRG31.Q3-23-27

conradalf, 2022-10-14:
Dir sir,
  Please release kernel source code for Motorola Edge (XT2143-1) RRG31.Q3-23-27 for android 11.
  Thanks a lot!!!

SyberHexen, 2022-11-02:
This is moto edge 20 (Berlin)
Request wants...
kernel-msm
kernel-devicetree
kernel-camera-devicetree
kernel-display-devicetree
kernel-msm-4.19-techpack-camera
kernel-msm-4.19-techpack-display
kernel-msm-4.19-techpack-video
vendor-cirrus-audio-dlkm
vendor-qcom-opensource-audio-kernel
vendor-qcom-opensource-wlan-qcacld-3.0
vendor-qcom-opensource-wlan-fw-api
vendor-qcom-opensource-wlan-qca-wifi-host-cmn

chenyt9, 2022-12-21:
@conradalf @SyberHexen , we will trigger the release of this version RRG31.Q3-23-27.
Here is a release for Motorola Edge 30：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRG31.Q3-23-85
Can you please use this version first？

chenyt9, 2023-01-19:
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRG31.Q3-23-27

closed 2023-01-19 by chenyt9