MotorolaMobilityLLC/kernel-msm #455: GPL | Source request for Motorola Edge 30 (dubai) (S1RDS32.55-94)

ElectroPerf, 2022-11-23:
Hey,

      Please do update the sources(kernel-msm, kernel-modules, techpacks, vendor, etc..) for the requested version S1RDS32.55-94 of Moto Edge 30(dubai).

sb6596, 2022-11-23:
@chenyt9 Please do the needful. Thank you

ThEMarD, 2022-11-23:
Hello, I would also like updated source code for Edge 30 too please

chenyt9, 2022-12-21:
@ElectroPerf ， I did not find this version S1RDS32.55-94, is the correct version number S1RD32.55-94?If yes, we will trigger the release of this version S1RD32.55-94 after getting the confirmation.
Here is a release for Motorola Edge 30：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1RD32.55-33
Can you please use this version first？

ThEMarD, 2022-12-21:
@chenyt9 sorry about that typo he had. You are correct, we need the source code for S1RD32.55-94 as there was changes in the device firmware which is causing loudspeaker to not work when we use the source code from S1RD32.55-33. This has been verified by a few dubai community developers.

Thank you,
Marc Bourgoin

chenyt9, 2022-12-21:
@ThEMarD ，
Ok, This version is already in our release process. Really appreciate your patience.

ThEMarD, 2022-12-22:
@chenyt9 at the moment I don't see S1RD32.55-94 and I just noticed that S1RDS32.55-94-2 is out now... is there a chance instead that the source code of tag S1RDS32.55-94-2 be released instead please?

ThEMarD, 2023-01-03:
@chenyt9 please any update on this? The new version hasn't been released yet. Can you please release S1RD32.55-94-2 kernel source code?

chenyt9, 2023-01-19:
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1RDS32.55-94-2

closed 2023-01-19 by chenyt9