MotorolaMobilityLLC/kernel-msm #457: GPL | Source request for Motorola Moto Edge 30 Neo (Miami) S3SSM32.34-13-3 (XT22245-1)

marcost2, 2022-12-01:
Kindly requesting source for the following repos for this tag "S3SSM32.34-13-3 "

kernel-msm
kernel-devicetree
kernel-camera-devicetree
kernel-display-devicetree
kernel-msm-5.4-techpack-camera
kernel-msm-5.4-techpack-display
kernel-msm-5.4-techpack-video
vendor-qcom-opensource-audio-kernel
vendor-qcom-opensource-wlan-qcacld-3.0
vendor-qcom-opensource-wlan-fw-api
vendor-qcom-opensource-wlan-qca-wifi-host-cmn

Thanks in advance!

chenyt9, 2022-12-21:
Hi,@marcost2 , could you use this version first:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SSM32.34-46-6
This version S3SSM32.34-46-1-1 will be released as soon as possible.

chenyt9, 2022-12-21:
@marcost2 ,Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SSMS32.34-46-1-1

closed 2022-12-21 by chenyt9

marcost2, 2022-12-21:
Hi, i'm still missing the following

kernel-msm-5.4-techpack-camera
kernel-msm-5.4-techpack-display
kernel-msm-5.4-techpack-video
vendor-qcom-opensource-audio-kernel
vendor-qcom-opensource-wlan-qcacld-3.0
vendor-qcom-opensource-wlan-fw-api
vendor-qcom-opensource-wlan-qca-wifi-host-cmn