MotorolaMobilityLLC/kernel-msm #458: [Request]GPL | Motorola Edge 30 Pro Source release (Device Tree, Kernal code and vendor etc.)

Mohali2609, 2022-12-06:
requesting source for the following repos:
 
- RETEU - S1SHS32.55-70-5-2
- RETEU - S3SH32.12-41-4

Thank you!

chenyt9, 2022-12-21:
@Mohali2609 ， I did not find this version S3SH32.12-41-4, is the correct version number S3SH32.12-41-4R1? If yes, we will trigger the release of this version S3SH32.12-41-4R1 after getting the confirmation.
Here is a release for Motorola Edge 30 Pro：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SH32.55-8-8
Can you please use this version first？

Mohali2609, 2022-12-22:
@chenyt9

The correct version number is S3SH32.12-41-4R1.

I will a look at the mentioned repo in the mean time.

Thanks in advance!

chenyt9, 2023-01-19:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3SH32.12-41-4R1

closed 2023-01-19 by chenyt9

chenyt9, 2023-01-20:
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SHS32.55-70-5-2