MotorolaMobilityLLC/kernel-msm #459: [Request]GPL | Motorola Moto G200 5G/Motorola Edge S30 Source release (Device Tree, Kernal code and vendor etc.)

ZhiBeiZhe, 2022-12-10:
RETEU S1RXS32.50-13-10

thanks

ZhiBeiZhe, 2022-12-10:
@awheeler-mmi@chenyt9@mbissaromoto

ThEMarD, 2022-12-22:
Heyyo @ZhiBeiZhe , whilst there isn't a direct source code release for xpeng, there is for tundra (Motorola Edge 30 Fusion/Motorola S30 Pro) which has the same SoC and the kernel source code contains the necessary parts for xpeng as well so in theory you should be able to use that tree.

Actually, even the source code release of dubai (Motorola Edge 30) has tundra and xpeng, so if you look at the LineageOS trees? You should even be able to adapt them for xpeng, albeit preferrably a newer source code release like tundra for the kernel to be used as there could be better camera drivers in there and maybe audio too.

You can see the initial tree adaptation I did here for tundra:

https://github.com/moto-sm7325/android_device_motorola_tundra/tree/lineage-20

https://github.com/moto-sm7325/android_device_motorola_sm7325-common/tree/tundra/lineage-20

https://github.com/moto-sm7325/android_kernel_motorola_sm7325/tree/tundra/lineage-20

https://gitlab.com/moto-sm7325/proprietary_vendor_motorola_tundra/-/tree/lineage-20

https://gitlab.com/moto-sm7325/proprietary_vendor_motorola_sm7325-common/-/tree/tundra/lineage-20

So you could fork those and modify them for xpeng to get started and then redo the blobs from the xpeng stock ROM to get your vendor blobs too.

The kernel probably also needs a few extra drivers setup to be buildable for Xpeng as well.

If you use Telegram? My username on there is ThE_MarD if you want to discuss this stuff further.

rogers2602, 2022-12-28:
@chenyt9 could you give a focus on releasing source code three of s30 Edge and g200? xpeng... thank you

dequadim, 2023-01-04:
Please, help us to give to our G200 a new life.
I strongly wish to give my help, but I'm a totally newbie and don't know where put my hands.

asbraga-motorola, 2023-02-13:
Code is released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1RXS32.50-13-10

closed 2023-02-13 by asbraga-motorola