MotorolaMobilityLLC/kernel-msm #460: GPL | Source request for Motorola  g200 (xpeng) (S1RXS32.50-13-10)

rogers2602, 2022-12-28:
Kindly requesting source for the following repos for this tag "S1RXS32.50-13-10"

kernel-msm
kernel-devicetree
kernel-audio-devicetree
kernel-camera-devicetree
kernel-cvp-devicetree
kernel-display-devicetree
kernel-mmrm-devicetree
kernel-video-devicetree
kernel-msm-5.4.147-techpack-camera
kernel-msm-5.4.147-techpack-cvp
kernel-msm-5.4.147-techpack-display
kernel-msm-5.4.147-techpack-eva
kernel-msm-5.4.147-techpack-mmrm
kernel-msm-5.4.147-techpack-sched
kernel-msm-5.4.147-techpack-video
vendor-qcom-opensource-audio-kernel
vendor-qcom-opensource-dataipa
vendor-qcom-opensource-datarmnet
vendor-qcom-opensource-datarmnet-ext
vendor-qcom-opensource-wlan-qcacld-3.0
vendor-qcom-opensource-wlan-fw-api
vendor-qcom-opensource-wlan-qca-wifi-host-cmn
motorola-kernel-modules

and others if applicable.

Thanks in advance!

dequadim, 2023-01-24:
No one cares about this phone 😓

closed 2023-02-13 by asbraga-motorola

rogers2602, 2023-02-14:
thank you very much

rogers2602, 2023-02-24:
@asbraga-motorola I tried to compile following the instructions in the readme that was released for xpeng, but I always get this error and I don't know what else to do, could you give me some light?

even the instructions there point to google's git directories of binaries that don't even exist anymore, to solve this I needed to do some git reverts to adapt the instructions.

the error i am having:

make[3]: *** No rule to process target 'arch/arm64/boot/dts/allwinner/sun50i-a64-amarula-relic.dtb', needed by '__build'. Stop.
make[2]: *** [/home/rogers/Documents/compilation/my_top_dir/kernel/msm-5.4/scripts/Makefile.build:558: arch/arm64/boot/dts/allwinner] Error 2
make[2]: ** Waiting for other processes to terminate.
make[1]: *** [/home/rogers/Documents/compilation/my_top_dir/kernel/msm-5.4/Makefile:1381: dtbs] Error 2
make[1]: ** Waiting for other processes to terminate.
   HOSTLD scripts/genksyms/genksyms
make[1]: Leaving directory '/home/rogers/Documentos/compilacao/my_top_dir/out/target/product/generic/obj/kernel/msm-5.4'
make: *** [Makefile:183: sub-make] Error 2
make: Leaving directory '/home/rogers/Documentos/compilacao/my_top_dir/kernel/msm-5.4'

rogers2602, 2023-03-01:
> @rogers2602 I reproduced the error, we are working to see what is wrong.

thank you, any news or update of the readme tag me here please thank you.

rogers2602, 2023-03-12:
> > @rogers2602 I reproduced the error, we are working to see what is wrong.
> 
> thank you, any news or update of the readme tag me here please thank you.

Any progress?

rogers2602, 2023-03-12:
> @rogers2602 I reproduced the error, we are working to see what is wrong.

Any progress?

rogers2602, 2023-03-25:
@asbraga-motorola @motorolamobility after studying the compilation codes of the drivers, I saw that there were some inconsistencies in the ALL winer and qcom drive paths, and I managed to correct them, After that I managed to compile the kernel and wlan, but I believe that the g200 readme still lacks some instructions to compile the wlan driver for the kernel, because in the log I don't see it adding the drive Qcalcl3.0 to the kernel, could you help me? I managed to test the kernel on the device but the network signal does not work on the mobile network in this case, other than the wi-fi.

asbraga-motorola, 2023-03-28:
Thanks, @rogers2602.
Can you share the modifications you made to the paths and readme thru a pull request? I'll compare them with the baseline to check what's wrong.

asbraga-motorola, 2023-04-06:
Hi @rogers2602,
For the **'No rule to process target'** error you reported above, it seems the problem is related to dtc.
You can clone it from https://android.googlesource.com/platform/prebuilts/misc to a local directory (e.g. $my_top_dir/prebuilts), create two variables like:

`local_dtc=$my_top_dir/prebuilts/misc/linux-x86/dtc/dtc`
`local_ufdt=$my_top_dir/prebuilts/misc/linux-x86/libufdt/ufdt_apply_overlay`

and replace all occurrences of the parameters DTC and DTC_OVERLAY_TEST_EXT in the readme:
`DTC_EXT=$local_dtc` and `DTC_OVERLAY_TEST_EXT=$local_ufdt`

No need to change anything in the code, only in the build instructions.
Can you check if you can compile the original code (as published) with these suggestions?

rogers2602, 2023-04-06:
@asbraga-motorola I'll try thanks for the feedback, a question I don't know if you'll know how to say it, do you know how I can compile the ramdisk, or make the final boot image without just the kernel? thank you if you can answer

rogers2602, 2023-04-09:
> Hi @rogers2602, For the **'No rule to process target'** error you reported above, it seems the problem is related to dtc. You can clone it from [...]

about the variables it worked, it compiled with some adjustments in the extract cert and signal, and it didn't give an error in the original code, thanks for your attention, if possible can you answer another question I asked?

asbraga-motorola, 2023-04-10:
Hi @rogers2602,
Good to know that it worked.
This forum is mainly for publication of OSS code, not for applications or support, so I don't have additional information for the other question. Maybe you could try the developers community: https://forums.lenovo.com/t5/MOTOROLA-Android-Developer/bd-p/230