MotorolaMobilityLLC/kernel-msm #467: Moto G200

dequadim, 2023-01-20:
Need for Moto G200 device tree.
Can someone help?

chenyt9, 2023-02-08:
Hi, @dequadim，
 Could you inform the exact released version you need?Thanks.

dequadim, 2023-02-08:
S1RXS32.50-13-12

Mauro Comiotto

Closed #467<https://github.com/MotorolaMobilityLLC/kernel-msm/issues/467> as completed.

rogers2602, 2023-02-13:
@chenyt9 please release the sources reported above.

metl-play, 2023-02-17:
I would like to have it too.

ThEMarD, 2023-05-16:
Heyyo @dequadim @metl-play @asbraga-motorola please see my local manifest here.

I got xpeng working on LineageOS and in this repo is also a README with instructions on how to set it up, sync and build for xpeng.

I do not own an xpeng so I cannot go official with LineageOS though...

https://github.com/moto-sm7325/local_manifests

asbraga-motorola, 2023-05-30:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1RXS32.50-13-12

closed 2023-05-30 by asbraga-motorola