MotorolaMobilityLLC/kernel-msm #468: Release Moto g62 global version

GabriLex, 2023-01-24:


asbraga-motorola, 2023-02-07:
Hi, @GabriLex 
Can you inform the exact released version you want?
You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2.

GabriLex, 2023-02-07:
> Hi, @GabriLex 
> Can you inform the exact released version you want?
> You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2.

Hi
I have S1SSS32.53-82-3 and codename rhodec
( I am european)

chenyt9, 2023-02-08:
@GabriLex ，
I've triggered the release process for version S1SSS32.53-82-3,  please be patient. Thanks!

chenyt9, 2023-03-06:
@GabriLex 
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S1SSS32.53-82-3
Thanks.

closed 2023-03-06 by chenyt9