MotorolaMobilityLLC/kernel-msm #473: [Request] (moto g8 plus (XT2019-1)) QPIS30.28-Q3-28-26-4-1-7

metl-play, 2023-02-17:
Hello,
I want to Request some files which are, namely but not limited to: Kernel Source, Device-Tree, "Vendor-Blobs"
(moto g8 plus (XT2019-1)) QPIS30.28-Q3-28-26-4-1-7 
Of course, only if possible.

metl-play, 2023-02-17:
Maybe, if existent, even unofficial updates to support Android 11 or newer.

asbraga-motorola, 2023-02-20:
Hi @metl-play 
I included QPIS30.28-Q3-28-26-4-1-7 in the pipeline for publication.
There is an earlier version already published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPIS30.28-Q3-28-26-4-1-4
For official upgrades, you can check https://motorola-global-portal.custhelp.com/app/software-upgrade

metl-play, 2023-02-20:
Thank you very much, and thanks for clarification about the already published version.

asbraga-motorola, 2023-04-06:
Hi @metl-play,
Code is published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPIS30.28-Q3-28-26-4-1-7

closed 2023-04-06 by asbraga-motorola