MotorolaMobilityLLC/kernel-msm #497: GPL | Device Tree and Kernel Source Code Request for Motorola Edge 20 Pro (pstar) T1RA33.55-15-10 (XT2153-1)

ericclose, 2023-04-22:
We kindly request the source code for this tag: T1RA33.55-15-10 on Android 13

kaloresanket, 2023-04-23:
@chenyt9 Kindly look into it and it would be heavily appreciated if you don't take a year to release half-baked sources as the trend has been so far.

JakoDel, 2023-05-01:
> @chenyt9 Kindly look into it and it would be heavily appreciated if you don't take a couple of months to release the sources as the trend has been so far.

what's with this almost aggressive tone? I know they are obliged to release the sources, but no one actually forces them to release source updates. be thankful guys come on.

asbraga-motorola, 2023-05-25:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1RA33.55-15-10

richard4O4, 2023-05-26:
> Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1RA33.55-15-10

@asbraga-motorola Could you please share the manual on how to build this kernel version for pstar? If you can, we would be very appreciated.

chenyt9, 2023-05-26:
@all, Pls refer to:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-T1RA33.55-15-10
Thanks.

kaloresanket, 2023-05-27:
@chenyt9 Thank you very much.
 I really appreciate the dedication and hard work Motorola is showing in hearing out and fixing the requirements of the customer base.

kaloresanket, 2023-05-27:
> > @chenyt9 Kindly look into it and it would be heavily appreciated if you don't take a couple of months to release the sources as the trend has been so far.
> 
> what's with this almost aggressive tone? I know they are obliged to release the sources, but no one actually forces them to release source updates. a few years ago the manufacturers releasing updated sources could probably be counted on one hand. you should be thankful.

ok dance then