MotorolaMobilityLLC/kernel-msm #500: GPL | Source request for Motorola Edge 40 Pro (rtwo) T1TR33.43-20-15

FuccDucc, 2023-05-07:
Please release the kernel tree, device tree and all the necessary audio/video/display techpack for Motorola Edge 40 (rtwo) version `T1TR33.43-20-15`

cc @chenyt9 @awheeler-mmi @mbissaromoto @asbraga-motorola @TinnoOSS

TAG=S1RDS32.55-106-3

FuccDucc, 2023-05-07:
I hope this isn't going to take months (or even lead to violating GPL laws, by indefinitely not being processed) as has seemingly been the trend over here. This is your newest device, it's good practise to release it off the bat, so development can begin during the moment of peak interest.

FuccDucc, 2023-05-12:
@asbraga-motorola i see you are online now, please notice this!

chenyt9, 2023-05-16:
@FuccDucc ，
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1TR33.43-20-15

closed 2023-05-16 by chenyt9

FuccDucc, 2023-05-16:
> @FuccDucc ， Version released： https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1TR33.43-20-15
@chenyt9 

Thanks, but please release the latest version too: `T1TR33.43-20-28`
[https://github.com/MotorolaMobilityLLC/kernel-msm/issues/503](https://github.com/MotorolaMobilityLLC/kernel-msm/issues/503) it was requested.