MotorolaMobilityLLC/kernel-msm #506: GPL | Source request | Motorola Edge 30 Pro / Motorola Edge Plus (2022) / Motorola Edge Plus 5G UW / Motorola Edge X30 / Motorola Edge X30: USC (hiphi/hiphic/hiphid)

7Soldier, 2023-05-21:
Dear Motorola Mobility, kindly requesting sources for HIPHI phones family:

Android 13:
- Motorola Edge 30 Pro - XT2201-1_HIPHI_RETEU_13_T1SH33.35-23-20 - HIPHI
- Motorola Edge Plus (2022) - XT2201-3_HIPHI_RETUS_13_T1SH33.35-23-20 - HIPHI
- Motorola Edge Plus 5G UW - XT2201-4_HIPHI_SPECTRUM_13_T1SH33.35-23-20 - HIPHI

Android 12:
- Motorola Edge 30 Pro - XT2201-1_HIPHI_RETEU_12_S3SHS32.12-41-4-5 - HIPHI
- Motorola Edge Plus (2022) - XT2201-3_HIPHI_RETUS_12_S3SHS32.12-41-4-5 - HIPHI
- Motorola Edge Plus 5G UW - XT2201-4_HIPHI_SPECTRUM_12_S3SHS32.12-41-4-1  /  XT2201-4_HIPHI_VZW_12_S3SHS32.12-41-4-5 - HIPHI
- Motorola Edge X30 - XT2201-2_HIPHIC_RETCN_12_S3SC32.7-42-20 - HIPHIC
- Motorola Edge X30 (USC) - XT2201-6_HIPHID_RETCN_12_S3SF32.35-5-9 - HIPHID

Including opensource repositories (for all hiphi/hiphic/hiphid models, different kernels or both in one sources):
- kernel-msm
- motorola-kernel-modules
- kernel-devicetree
- kernel-audio-devicetree
- kernel-camera-devicetree
- kernel-cvp-devicetree
- kernel-display-devicetree
- kernel-mmrm-devicetree
- kernel-video-devicetree
- kernel-msm-5.10-techpack-camera
- kernel-msm-5.10-techpack-cvp
- kernel-msm-5.10-techpack-display
- kernel-msm-5.10-techpack-eva
- kernel-msm-5.10-techpack-mmrm
- kernel-msm-5.10-techpack-sched
- kernel-msm-5.10-techpack-video
- vendor-qcom-opensource-audio-kernel
- vendor-qcom-opensource-dataipa
- vendor-qcom-opensource-datarmnet
- vendor-qcom-opensource-datarmnet-ext
- vendor-qcom-opensource-wlan-qcacld-3.0
- vendor-qcom-opensource-wlan-fw-api
- vendor-qcom-opensource-wlan-qca-wifi-host-cmn
- and others

Thanks in advice, waiting for respond!
Sincerely, 7Soldier

CC: @chenyt9 @awheeler-mmi @asbraga-motorola @TinnoOSS @mbissaromoto

s4704, 2023-05-21:
Please post the full source code.

LucyfMorning, 2023-05-21:
Yeah..  release full code

Anush02198, 2023-05-21:
Please Release Source code .. @motorolamobility @awheeler-mmi @asbraga-motorola

7Soldier, 2023-05-25:
Please, don't ignore our requests for sources.
We need all mentioned repositories source code at least for T1SH33.35-23-20.

Thanks in advice!

CC:
@chenyt9
@awheeler-mmi
@asbraga-motorola
@motorolamobility
@jwylder1
@mbissaromoto
@ccfries

7Soldier, 2023-06-27:
Upping issue.

CC:
@chenyt9
@awheeler-mmi
@asbraga-motorola
@motorolamobility
@jwylder1
@mbissaromoto
@ccfries
@TinnoOSS

YeYdi-YonY, 2023-06-28:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SH33.35-23-20

closed 2023-07-13 by 7Soldier