MotorolaMobilityLLC/kernel-msm #508: GPL | Source request for Motorola Moto G42 (Hawao) [S2SES32.28-67-6] Android 12 Sources release (Device Tree, Kernel code, etc.)

MiniViper111, 2023-05-23:
Hello,
I would like to request the release of files such as kernel tree, device tree, vendor-Blobs and  all the necessary files for Moto G42 (Hawao) [S2SES32.28-67-6] Android 12.

cc @chenyt9 @awheeler-mmi @mbissaromoto @asbraga-motorola @TinnoOSS

MiniViper111, 2023-06-21:
Requesting again to release the device tree, source code.

MiniViper111, 2023-07-05:
Requesting again to please release the kernel tree, vendor, device tree and all the necessary audio/video/display techpack for Motorola motog42

CC:
@chenyt9
@awheeler-mmi
@asbraga-motorola
@motorolamobility
@jwylder1
@mbissaromoto
@ccfries
@TinnoOSS

chenyt9, 2023-07-27:
@MiniViper111 
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S2SES32.28-67-7
Sorry for delay.

closed 2023-07-27 by chenyt9

MiniViper111, 2023-07-28:
Thanks @chenyt9