MotorolaMobilityLLC/kernel-msm #516: GPL | Source request for Motorola Edge 40 Pro (rtwo) XT2301-4 - T1TR33.43-20-28

SGCMarkus, 2023-06-03:
Could you please release the sources for the following repositories (based on current naming scheme)?
kernel-audio-devicetree
kernel-bt-devicetree
kernel-camera-devicetree
kernel-data-devicetree
kernel-data-devicetree
kernel-data-devicetree
kernel-data-devicetree
kernel-data-devicetree
kernel-data-devicetree
kernel-data-devicetree

FuccDucc, 2023-06-11:
You forgot to add to your list:

kernel-msm-techpack-display
kernel-msm-techpack-video
kernel-msm-techpack-dataipa

Like #509 had included

@SGCMarkus can edit it in, or else i request that who process this will take it along.

FuccDucc, 2023-06-11:
For the entire issue:

cc @chenyt9 @awheeler-mmi @mbissaromoto @asbraga-motorola @TinnoOSS

chenyt9, 2023-07-14:
@SGCMarkus 
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1TR33.43-20-28

closed 2023-07-14 by chenyt9