MotorolaMobilityLLC/kernel-msm #522: GPL | Source request | Motorola Edge 30 Pro (hiphi/hiphic/hiphid)

7Soldier, 2023-06-15:
Dear Motorola Mobility, kindly requesting last sources for HIPHI phones family:
- T1SH33.35-23-20-2
- S3SHS32.12-41-4-5
- S3SF32.35-5-9

Including opensource repositories:
- kernel-msm
- motorola-kernel-modules
- kernel-devicetree
- kernel-audio-devicetree
- kernel-bt-devicetree
- kernel-camera-devicetree
- kernel-data-devicetree
- kernel-display-devicetree
- kernel-eSE-devicetree
- kernel-eva-devicetree
- kernel-graphics-devicetree
- kernel-mm-devicetree
- kernel-mmrm-devicetree
- kernel-nfc-devicetree
- kernel-video-devicetree
- kernel-wlan-devicetree
- kernel-msm-5.10-techpack-camera
- kernel-msm-5.10-techpack-cvp
- kernel-msm-5.10-techpack-display
- kernel-msm-5.10-techpack-eva
- kernel-msm-5.10-techpack-mmrm
- kernel-msm-5.10-techpack-sched
- kernel-msm-5.10-techpack-video
- vendor-qcom-opensource-audio-kernel
- vendor-qcom-opensource-dataipa
- vendor-qcom-opensource-datarmnet
- vendor-qcom-opensource-datarmnet-ext
- vendor-qcom-opensource-wlan-qcacld-3.0
- vendor-qcom-opensource-wlan-fw-api
- vendor-qcom-opensource-wlan-qca-wifi-host-cmn
- and others required for working stuff

Thanks in advice, waiting for respond!
Sincerely, 7Soldier

CC:
@chenyt9
@awheeler-mmi
@asbraga-motorola
@motorolamobility
@jwylder1
@mbissaromoto
@ccfries
@TinnoOSS

SunsetTechuila, 2023-06-19:
no

7Soldier, 2023-06-27:
Upping issue.

Dear Motorola Mobility, kindly requesting last sources for hiphi/hiphic/hiphid phones family:
- `T1SC33.52-8-6`
- `S3SHS32.12-41-4-5`
- `S3SF32.35-5-9`

Including opensource repositories:
- kernel-msm
- motorola-kernel-modules
- kernel-devicetree
- kernel-audio-devicetree
- kernel-camera-devicetree
- kernel-display-devicetree
- kernel-mmrm-devicetree
- kernel-video-devicetree
- vendor-qcom-opensource-audio-kernel
- vendor-qcom-opensource-camera-kernel
- vendor-qcom-opensource-datarmnet
- vendor-qcom-opensource-datarmnet-ext
- vendor-qcom-opensource-eva-kernel
- vendor-qcom-opensource-mmrm-driver
- vendor-qcom-opensource-wlan-fw-api
- vendor-qcom-opensource-wlan-qca-wifi-host-cmn
- vendor-qcom-opensource-qcacld-3.0
- kernel-msm-5.10-techpack-cvp
- kernel-msm-5.10-techpack-display
- kernel-msm-5.10-techpack-dataipa
- kernel-msm-5.10-techpack-video
- and others required for working stuff (device tree, all the necessary audio/video/display techpack, etc.)

Thanks in advice, waiting for respond!
Sincerely, 7Soldier

CC:
@chenyt9
@awheeler-mmi
@asbraga-motorola
@motorolamobility
@jwylder1
@mbissaromoto
@ccfries
@TinnoOSS

SunsetTechuila, 2023-06-27:
No

YeYdi-YonY, 2023-06-28:
Hiphi:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SH33.35-23-20

7Soldier, 2023-06-28:
> Hiphi: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SH33.35-23-20

I am asking for hiphi a12 last sources, hiphic a13 sources, and hiphid a12 last sources. We need all branches to make one generic kernel.

chenyt9, 2023-07-12:
@all,
T1SC33.52-8-6 version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SC33.52-8-6

closed 2023-07-13 by 7Soldier