MotorolaMobilityLLC/kernel-msm #525: GPL | Source request for Motorola Edge X30 (hiphic) (XT2201-2) - T1SC33.52-8-6

7Soldier, 2023-06-27:
Dear Motorola Mobility, kindly requesting last sources for T1SC33.52-8-6:
- Model number: `XT2201-2`
- Codename: `hiphic`
- Version: `T1SC33.52-8-6`

Including all opensource repositories:
- kernel-msm
- motorola-kernel-modules
- kernel-devicetree
- kernel-audio-devicetree
- kernel-camera-devicetree
- kernel-display-devicetree
- kernel-mmrm-devicetree
- kernel-video-devicetree
- vendor-qcom-opensource-audio-kernel
- vendor-qcom-opensource-camera-kernel
- vendor-qcom-opensource-datarmnet
- vendor-qcom-opensource-datarmnet-ext
- vendor-qcom-opensource-eva-kernel
- vendor-qcom-opensource-mmrm-driver
- vendor-qcom-opensource-wlan-fw-api
- vendor-qcom-opensource-wlan-qca-wifi-host-cmn
- vendor-qcom-opensource-qcacld-3.0
- kernel-msm-5.10-techpack-cvp
- kernel-msm-5.10-techpack-display
- kernel-msm-5.10-techpack-dataipa
- kernel-msm-5.10-techpack-video
- and others required for working stuff (device tree, all the necessary audio/video/display techpack, etc.)

Thanks in advice, waiting for respond!
Sincerely, 7Soldier

CC:
@chenyt9
@awheeler-mmi
@asbraga-motorola
@motorolamobility
@jwylder1
@mbissaromoto
@ccfries
@TinnoOSS

SunsetTechuila, 2023-06-29:
No

chenyt9, 2023-07-12:
@7Soldier @SunsetTechuila ，
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SC33.52-8-6

closed 2023-07-12 by chenyt9