MotorolaMobilityLLC/kernel-msm #528: GPL | Kernel display drivers source request for Motorola Edge 40 Pro (rtwo) XT2301-4 - T1TR33.43-20-28

SGCMarkus, 2023-07-20:
Hi,

we are still missing the sources for the **kernel display drivers** ([kernel-msm-techpack-display](https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-display) / [kernel-msm-5.4-techpack-display](https://github.com/MotorolaMobilityLLC/kernel-msm-5.4-techpack-display)) for rtwo.
Could you upload them?

Thanks in advance

ThEMarD, 2023-07-20:
I would also like the missing sources for kernel display drivers

chenyt9, 2023-07-21:
@SGCMarkus 
Is this what you need?
https://github.com/MotorolaMobilityLLC/kernel-display-devicetree/releases/tag/MMI-T1TR33.43-20-28

SGCMarkus, 2023-07-21:
> @SGCMarkus Is this what you need? https://github.com/MotorolaMobilityLLC/kernel-display-devicetree/releases/tag/MMI-T1TR33.43-20-28

sadly no, thats the display devicetree
I would need the kernel display drivers (on older platforms techpack/display, i linked the repositories from here that you used in the past in the first post)
its similar to your vendor-qcom-opensource-camera-kernel or vendor-qcom-opensource-audio-kernel release, just instead of camera-kernel/audio-kernel its display-drivers.
-> vendor-qcom-opensource-display-drivers
or kernel-msm-techpack-display

7Soldier, 2023-07-22:
Same request for Motorola Edge 30 Pro (hiphi) - T1SH33.35-23-20.
Display techpack also wasn't released, so kindly requesting it.

FuccDucc, 2023-07-29:
@chenyt9 @awheeler-mmi @mbissaromoto @asbraga-motorola @TinnoOSS u see it?

XeonDead, 2023-08-03:
@SGCMarkus @ThEMarD those should 'probably' work: https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-display/tree/android-13-release-tsjc
Check ones prebuilt against this branch (most recent one, probably shouldn't hurt).

chenyt9, 2023-08-14:
@SGCMarkus ,
Is this what you need?
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-display-drivers/releases/tag/MMI-T1TR33.43-20-28

chenyt9, 2024-01-18:
@SGCMarkus ，
I will close this issue first. If you still have questions, please open another issue. Thank you!

closed 2024-01-18 by chenyt9

SGCMarkus, 2024-01-18:
> @SGCMarkus , Is this what you need? https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-display-drivers/releases/tag/MMI-T1TR33.43-20-28

> @SGCMarkus ， I will close this issue first. If you still have questions, please open another issue. Thank you!

Sorry, for the (very) late reply. Yes, that repo you uploaded on the 14. August was the one we needed.