MotorolaMobilityLLC/kernel-msm #531: [REQUEST] Kernel Release for RPXS31.Q2-58-17-7-3 guamp

RazaDroid, 2023-07-25:
RPXS31.Q2-58-17-7-3 has been published in September 2022, why kernel source not pushed yet?

@mbissaromoto

asbraga-motorola, 2023-09-04:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RPXS31.Q2-58-17-7-3

RazaDroid, 2023-09-04:
@asbraga-motorola other repos are not pushed device tree, kernel modules, techpack and all..

asbraga-motorola, 2023-09-08:
I'll check why they are missing.

asbraga-motorola, 2023-09-12:
The missing repos are published now. You may look for the tag MMI-RPXS31.Q2-58-17-7-3.

closed 2023-09-12 by asbraga-motorola

gelomon, 2023-10-21:
Hello as per checking the tag it is still missing

asbraga-motorola, 2023-10-23:
The tag MMI-RPXS31.Q2-58-17-7-3 is applied to the repos

- kernel-camera-devicetree
- kernel-devicetree
- kernel-display-devicetree
- kernel-msm
- kernel-msm-techpack-camera
- kernel-msm-techpack-display
- kernel-msm-techpack-video
- vendor-cirrus-audio-dlkm
- vendor-qcom-opensource-audio-kernel
- vendor-qcom-opensource-data-kernel
- vendor-qcom-opensource-wlan-fw-api
- vendor-qcom-opensource-wlan-qca-wifi-host-cmn
- vendor-qcom-opensource-wlan-qcacld-3.0

and many others.
Could you please check again and specify which ones are missing?

gelomon, 2023-11-04:
Thank you @asbraga-motorola. I have rechecked some of them and github is not showing them until I searched the whole tag. I will be verifying them.

Regards