MotorolaMobilityLLC/kernel-msm #553: [REQUEST] Kernel Source for Motorola G52j 5G (cypfr) - S3RYBS32.168-19-5-11

Garry050, 2023-10-26:
[REQUEST] Kernel Source for Motorola G52j 5G (cypfr) - S3RYBS32.168-19-5-11

Please release the kernel tree, device tree and all the necessary audio/video/display techpack for Motorola G52j 5G (cypfr) version S3RYBS32.168-19-5-11

Model number: XT2219-1
Codename: cypfr

Remind: This is not Global model. only Japan retail model
<Kernel version has been updated to 5.4.210 - 5.4.233>

CC: @chenyt9 @awheeler-mmi @TinnoOSS @asbraga-motorola @mbissaromoto

asbraga-motorola, 2023-11-30:
code released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S3RYBS32.168-19-5-11

closed 2023-11-30 by asbraga-motorola

Garry050, 2023-12-09:
Thank you, But 1 missing techpack.
https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-dataipa/tags

CC: @chenyt9 @awheeler-mmi @TinnoOSS @asbraga-motorola @mbissaromoto

acethical, 2024-04-11:
> Thank you, But 1 missing techpack. https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-dataipa/tags

There's a tag in the repo you mentioned:
https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-dataipa/releases/tag/MMI-S3RYBS32.168-19-5-3
That seems to be the same device's techpack, except the version are different.
 Not sure it's helpful or not.

asbraga-motorola, 2024-04-26:
I created a new tag: https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-dataipa/releases/tag/MMI-S3RYBS32.168-19-5-11