MotorolaMobilityLLC/kernel-msm #556: [REQUEST] Latest troika source code

npjohnson, 2023-10-30:
Please release the kernel tree, device tree and all the necessary audio/video/display techpack for Motorola Edge 30 Pro version TROIKA_RETAIL_11_RSBS31.Q1-48-36-26_subsidy-DEFAULT_regulatory-DEFAULT_CFC.xml.zip

Codename: troika
Version: RSBS31.Q1-48-36-26
Variant: RETAIL

While processing this request, please take care to include all of the repositories that correspond to ones such as mentioned in https://github.com/MotorolaMobilityLLC/kernel-msm/issues/516 (referenced just as example of what sort of repo's from the tree appear to be easy to miss during a GPL release)

Notif: @chenyt9 @awheeler-mmi @mbissaromoto @asbraga-motorola @TinnoOSS

TAG=RSBS31.Q1-48-36-26

npjohnson, 2023-12-11:
whoops, wrong kernel repo

closed 2023-12-11 by npjohnson

asbraga-motorola, 2023-12-13:
Code published: https://github.com/MotorolaMobilityLLC/kernel-slsi/releases/tag/MMI-RSBS31.Q1-48-36-26