MotorolaMobilityLLC/kernel-msm #561: [Request] Which defconfig files needed to build kernel for moto g53 5G (XT2335-1; retbr; holi; penang)?

D4yvid, 2023-11-18:
How i can build a kernel for the moto g53 5G? I found it's [kernel source tag](https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-T1TP33.1-47-2), but there is little to no information on which defconfigs i should use, nor how to build it correctly.

I tried building this kernel source tag, using this build configuration:
LLVM toolchain: https://github.com/LineageOS/android_prebuilts_clang_kernel_linux-x86_clang-r416183b
GCC/Binutils for aarch64: https://github.com/LineageOS/android_prebuilts_gcc_linux-x86_aarch64_aarch64-linux-android-4.9
GCC/Binutils for arm32: https://github.com/LineageOS/android_prebuilts_gcc_linux-x86_arm_arm-linux-androideabi-4.9

<details>
  <summary>Output<pre>d4y@moon:~/kernel-msm$ ./scripts/kconfig/merge_config.sh -m arch/arm64/configs/vendor/genericarmv8-64_defconfig arch/arm64/configs/vendor/holi_debug.config arch/arm64/configs/vendor/ext_config/debug-holi.config arch/arm64/configs/vendor/ext_config/debug-holi-penang.config</pre></summary>
  <pre>
Using arch/arm64/configs/vendor/genericarmv8-64_defconfig as base
Merging arch/arm64/configs/vendor/holi_debug.config
Value of CONFIG_DEBUG_PREEMPT is redefined by fragment arch/arm64/configs/vendor/holi_debug.config:
Previous value: # CONFIG_DEBUG_PREEMPT is not set
New value: CONFIG_DEBUG_PREEMPT=y

Merging arch/arm64/configs/vendor/ext_config/debug-holi.config
Value of CONFIG_EDAC_KRYO_ARM64_PANIC_ON_CE is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_EDAC_KRYO_ARM64_PANIC_ON_CE=y
New value: # CONFIG_EDAC_KRYO_ARM64_PANIC_ON_CE is not set

Value of CONFIG_SLUB_DEBUG_PANIC_ON is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_SLUB_DEBUG_PANIC_ON=y
New value: # CONFIG_SLUB_DEBUG_PANIC_ON is not set

Value of CONFIG_DEBUG_PAGEALLOC is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_DEBUG_PAGEALLOC=y
New value: # CONFIG_DEBUG_PAGEALLOC is not set # CONFIG_DEBUG_PAGEALLOC is not set

Value of CONFIG_DEBUG_PAGEALLOC_ENABLE_DEFAULT is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_DEBUG_PAGEALLOC_ENABLE_DEFAULT=y
New value: # CONFIG_DEBUG_PAGEALLOC_ENABLE_DEFAULT is not set # CONFIG_DEBUG_PAGEALLOC_ENABLE_DEFAULT is not set

Value of CONFIG_DEBUG_PANIC_ON_OOM is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_DEBUG_PANIC_ON_OOM=y
New value: # CONFIG_DEBUG_PANIC_ON_OOM is not set

Value of CONFIG_DEBUG_PREEMPT is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_DEBUG_PREEMPT=y
New value: # CONFIG_DEBUG_PREEMPT is not set

Value of CONFIG_TOUCHSCREEN_FTS is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_TOUCHSCREEN_FTS=y
New value: # CONFIG_TOUCHSCREEN_FTS is not set

Value of CONFIG_CORESIGHT_CTI_SAVE_DISABLE is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: # CONFIG_CORESIGHT_CTI_SAVE_DISABLE is not set
New value: CONFIG_CORESIGHT_CTI_SAVE_DISABLE=y

Value of CONFIG_DEBUG_SPINLOCK is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_DEBUG_SPINLOCK=y CONFIG_DEBUG_SPINLOCK=y
New value: CONFIG_DEBUG_SPINLOCK=y

Value of CONFIG_PREEMPT_TRACER is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_PREEMPT_TRACER=y
New value: # CONFIG_PREEMPT_TRACER is not set

Value of CONFIG_LOCK_STAT is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi.config:
Previous value: CONFIG_LOCK_STAT=y
New value: # CONFIG_LOCK_STAT is not set

Merging arch/arm64/configs/vendor/ext_config/debug-holi-penang.config
Value of CONFIG_CMA_ALLOW_WRITE_DEBUGFS is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi-penang.config:
Previous value: CONFIG_CMA_ALLOW_WRITE_DEBUGFS=y
New value: # CONFIG_CMA_ALLOW_WRITE_DEBUGFS is not set

Value of CONFIG_CMA_DEBUG is redefined by fragment arch/arm64/configs/vendor/ext_config/debug-holi-penang.config:
Previous value: CONFIG_CMA_DEBUG=y
New value: # CONFIG_CMA_DEBUG is not set

[... 105 lines cut ...]
  CC      drivers/irqchip/qcom-mpm-qcs405.o
  CC [M]  drivers/md/dm-stripe.o
  CC      drivers/irqchip/qcom-mpm-blair.o
  CC [M]  drivers/md/dm-ioctl.o
  CC [M]  drivers/md/dm-io.o
  CC [M]  drivers/md/dm-kcopyd.o
  AR      drivers/irqchip/built-in.a
  CC [M]  drivers/md/dm-sysfs.o
  CC [M]  drivers/md/dm-stats.o
  CC [M]  drivers/md/dm-rq.o
  CC [M]  drivers/md/dm-raid1.o
  CC [M]  drivers/md/dm-log.o
  CC [M]  drivers/md/dm-region-hash.o
  CC [M]  drivers/md/dm-zero.o
  AR      drivers/md/built-in.a
  LD [M]  drivers/md/dm-mod.o
  LD [M]  drivers/md/dm-mirror.o
  LD [M]  drivers/md/md-mod.o
make[1]: *** [/home/day/kernel-msm/Makefile:1845: drivers] Error 2
make[1]: Leaving directory '/home/day/kernel-msm/out'
make: *** [Makefile:183: sub-make] Error 2
</pre>
</details>

I really don't know which files i should use to build this, i did a lot of research on that but i wasn't able to solve this by myself.

chankruze, 2023-11-23:
@D4yvid Can you share the AIDA64 report for XT2335-1 ?

D4yvid, 2023-11-23:
> @D4yvid Can you share the AIDA64 report for XT2335-1 ?

Yeah! Here it is: https://pastebin.com/BnDiBS4U

chankruze, 2023-11-23:
Which andorid version you are building ?

Found that thing in `usb_bypass_notifier` function [here](https://android.googlesource.com/kernel/msm.git/+/a7b4e9f4453bc9f6ec8b6ae702d963d3f35800ca/drivers/hwtracing/coresight/coresight-byte-cntr.c) it no longer exit in android 13 ([here](https://android.googlesource.com/kernel/msm.git/+/refs/tags/android-13.0.0_r0.130/drivers/hwtracing/coresight/coresight-byte-cntr.c)).

D4yvid, 2023-11-23:
> Which andorid version you are building ? I think your config is correct.
> 
> Found that thing in `usb_bypass_notifier` function [here](https://android.googlesource.com/kernel/msm.git/+/a7b4e9f4453bc9f6ec8b6ae702d963d3f35800ca/drivers/hwtracing/coresight/coresight-byte-cntr.c) it no longer exit in android 13 ([here](https://android.googlesource.com/kernel/msm.git/+/refs/tags/android-13.0.0_r0.130/drivers/hwtracing/coresight/coresight-byte-cntr.c)).

I'm trying to build only the kernel, but the kernel is being built for Android 13

D4yvid, 2023-11-23:
Also, my phone is rooted, so i got the kernel config file from /proc/config.gz (the builders forgot to disable it lol), i'll try to build this again when i get home using this config file from /proc

Thank you for helping me c:

D4yvid, 2023-11-24:
@chankruze The kernel compiled successfully using the `/proc/config.gz` extracted directly from the device, thank you for helping me! I'll create a fork with this device config in my github in case someone need this to build.

closed 2023-11-24 by D4yvid