MotorolaMobilityLLC/kernel-msm #565: GPL - Source Request for Motorola g84 XT2347-2

chankruze, 2023-11-23:
Please release the kernel tree, device tree and all the necessary audio/video/display techpack for Motorola Moto g84 version `T3TC33.18-48-6`

| Key             | Value                                                |
| :-------------- | :--------------------------------------------------- |
| Model           | XT2347-2                                             |
| Codename        | REL                                                  |
| Build ID | T3TC33.18-48-6                                       |
| Software Channel         | retin                                               |
| Android Version | 13 (Tiramisu)                                        |
| API Level       | 33                                                   |
| Bootloader      | MBM-3.0-uefi-e8ce7c27e72-230928-T3TC33.18-48-6-5dba7 |

Can I use [MMI-T3TC33.18-12-3](https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-T3TC33.18-12-3) asking because the phone was shipped with `T3TC33.18-27-34`

chenyt9, 2023-12-14:
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T3TC33.18-48-6

closed 2023-12-14 by chenyt9

asbraga-motorola, 2025-08-06:
you can just open a new issue requesting the Build ID you want.