MotorolaMobilityLLC/kernel-msm #570: longer shutter speed for third party apps

Artem7shenin, 2023-11-26:
Can you tell me if there is a way to increase the shutter speed for third-party applications (currently it is 1/4) to, for example, 500 seconds or more.  for example by changing values in camera files or using moto camera tuner
Smartphone: Motorola edge 20, android 12 global rom

asbraga-motorola, 2023-12-13:
This forum is primarily for requesting open source code publication.
You may try the Moto Community Forum for that kind of question:
[https://forums.lenovo.com/t5/Motorola-Community/ct-p/MotorolaCommunity]

closed 2023-12-27 by asbraga-motorola

Artem7shenin, 2023-12-31:
I know, on the forum I received a vague answer - https://forums.lenovo.com/t5/motorola-edge-20/how-to-unlock-shutter-speed-more-than-1-4-in-third-party-camera-apps/m-p/5268333?page=1#6165386 , but I was hoping that people close to the source code could help me, in theory, if you know where the maximum shutter speed value is for third-party applications, then it can be quickly changed to the desired value (I  I would like 6000 seconds) and update the application in which this file is stored.  I have not found this meaning for a very long time and not only in this thread.  If there are any ways to implement this, I'm ready for anything

Artem7shenin, 2023-12-31:
> This forum is primarily for requesting open source code publication. You may try the Moto Community Forum for that kind of question: [https://forums.lenovo.com/t5/Motorola-Community/ct-p/MotorolaCommunity]

> This forum is primarily for requesting open source code publication. You may try the Moto Community Forum for that kind of question: [https://forums.lenovo.com/t5/Motorola-Community/ct-p/MotorolaCommunity]

I know, on the forum I received a vague answer - https://forums.lenovo.com/t5/motorola-edge-20/how-to-unlock-shutter-speed-more-than-1-4-in-third-party-camera-apps/m-p/5268333?page=1#6165386 , but I was hoping that people close to the source code could help me, in theory, if you know where the maximum shutter speed value is for third-party applications, then it can be quickly changed to the desired value (I I would like 6000 seconds) and update the application in which this file is stored. I have not found this meaning for a very long time and not only in this thread. If there are any ways to implement this, I'm ready for anything