MotorolaMobilityLLC/kernel-msm #576: GPL | Source request for Motorola Defy 2021 (XT2083-9) - RZD31.31

mihaimoiseanu, 2023-12-06:
Please release the kernel tree, device tree and all the necessary audio/video/display techpack for Motorola Defy 2021 version RZD31.31

Model number: XT2083-9
Version: RZD31.31 (March 2023)

asbraga-motorola, 2023-12-11:
RZD31.31 is not a released version. Can you double check that and inform the exact released version (Build ID) you want?
You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2, T2TZ33.28-16-19.

mihaimoiseanu, 2023-12-14:
Running `adb shell getprop` I found the following lines regarding the build id
[ro.product.build.id]: [RZD31.31]
[ro.build.id]: [RZD31.31]
[ro.system_ext.build.id]: [RZD31.31]
[ro.mot.base_buildid]: [RZD31.17/7dcbe9]

asbraga-motorola, 2024-01-17:
working on it.

asbraga-motorola, 2024-01-29:
Code released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RZD31.31

closed 2024-01-29 by asbraga-motorola