MotorolaMobilityLLC/kernel-msm #577: [GPL Request] Kernel source for motorola edge 30 fuision(tundra_g) T1SJS33.117-30-3-5

schqiushui, 2023-12-14:
Please release the kernel sources along with the usual device-trees, techpacks, modules, etc.
Thanks in advance.
@asbraga-motorola @chenyt9

asbraga-motorola, 2023-12-17:
Code released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SJS33.117-30-3-5

schqiushui, 2023-12-17:
> Code released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SJS33.117-30-3-5

Thanks a lot.

schqiushui, 2023-12-17:
> Code released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SJS33.117-30-3-5

Hi, I found out that missing some below, please help release these sources.
Thanks!

https://github.com/MotorolaMobilityLLC/vendor-cirrus-audio-dlkm
https://github.com/MotorolaMobilityLLC/kernel-devicetree
https://github.com/MotorolaMobilityLLC/kernel-msm-techpack-dataipa
https://github.com/MotorolaMobilityLLC/kernel-msm-5.4-techpack-camera
https://github.com/MotorolaMobilityLLC/kernel-msm-5.4-techpack-video

asbraga-motorola, 2024-01-09:
Fixed. Now those repos have the tag for this release.

closed 2024-01-09 by asbraga-motorola

schqiushui, 2024-01-10:
Thanks a lot.