MotorolaMobilityLLC/kernel-msm #581: Kernel Source Code request for XT2313 (pnangn).

DragonFire1024, 2023-12-21:
Would like to request the kernel source code for the Motorola Moto G 5G, 2023 (XT2313, pnangn).

DragonFire1024, 2023-12-22:
As I understand it, Penang is the 'international' variant whereas Pnangn is the 'North American' variant. These devices are the same when it comes to software/hardware? From what I looked at briefly, it suggests so. Would the kernel source from Penang be compatible with Pnangn?

asbraga-motorola, 2024-01-03:
Do you have a specific Build ID of pnangn?
You find it in the System Config in the phone. It is a group of letters and numbers that look like these: RRA31.Q3-19-19, S1RD32.55-33, S0RCS32.41-10-9-2, T2TZ33.28-16-19.

DragonFire1024, 2024-01-03:
ro.build.description: "T1TPN33.58-68-1-2-1 08053 release-keys"

chenyt9, 2024-01-18:
Version here：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1TPN33.58-68-1-2-1

closed 2024-01-18 by chenyt9