MotorolaMobilityLLC/kernel-msm #589: Correct Toolchain for Denver

pauldmurphy, 2024-01-20:
### Build ID

S2RE32.29-16-1-5-19 

### Additional

I'm having a nightmare of a time identifying the correct toolchain to build this kernel. Can you please advise?

Thank You!

--pm

pauldmurphy, 2024-01-31:
i found it here 

https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86/+/e532342f30eadb954c96d19ca6a0edf89a5c65bc/clang-r383902b1/

can confirm stock device config builds

metl-play, 2024-02-01:
How did you find the correct toolchain? I want to compile the kernel for my Doha device, unfortunately I am completely new to it.

pauldmurphy, 2024-02-04:
Here's how I found it.

This will work on almost all android devices. You will need root in order
to be able to perform these operations.

cp /proc/config.gz ~

gunzip ~/config.gz
or
gzip ~/config.gz

Rename that file to something more manageable. My version of that file had
a super long name. I renamed it to device_config

mv ~/[super long filename] ~/device_config

Open the file with your preferred editor. I'm keen on nano.

nano ~/device_config

The first couple of lines should point you into the right place for the
toolchain. The file itself is the .config file used to build the running
kernel.

HTH,
--pm

metl-play, 2024-02-06:
So I need to reinstall the stock firmware and then root it, to get this information?

juic3b0x, 2024-02-13:
> So I need to reinstall the stock firmware and then root it, to get this information?

or if you got boot.img just run:
`strings boot.img | grep "Linux version"`

And it'll show you the clang version also...

*quietly goes back to lurking*

closed 2024-05-09 by asbraga-motorola