MotorolaMobilityLLC/kernel-msm #594: Kernel Source for motorola edge 30 neo (miami) - T1SSMS33.1-121-4-8

Ev3rt, 2024-02-26:
### Build ID

T1SSMS33.1-121-4-8

asbraga-motorola, 2024-03-01:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SSMS33.1-121-4-8

closed 2024-03-01 by asbraga-motorola

AMT-Cheif, 2024-05-02:
> Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SSMS33.1-121-4-8

Can you upload device and vendor tree?

AMT-Cheif, 2024-05-06:
I founded the tag MMI-T1SSMS33.1-121-4-8 in almost all repository, but in:
vendor-qcom-opensource-wlan-fw-api I found only MMI-T1SSMS33.1-121-4-5 or MMI-T1SSMS33.1-121-4. Wich is the correct one?

asbraga-motorola, 2024-05-06:
It's there: https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-wlan-fw-api/releases/tag/MMI-T1SSMS33.1-121-4-8