MotorolaMobilityLLC/kernel-msm #596: [GPL Request] Kernel Source for T1SUS33.1-124-6-7 (Moto G82 5G / rhodep / XT2225-1)

sevenrock, 2024-03-19:
### Build ID

Hello,

could you please release the latest sources for the Kernel (and usual techpacks, modules, device trees, etc)
for the Motorola Moto G82 5G (rhodep / XT2225-1). The build number is T1SUS33.1-124-6-7.

The latest released source I could find for rhodep is for Android 12: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/S1SUS32.73-13-4-3
I could also find Android 13 tags for rhode ( https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T2SRS33.72-22-4-1 ) and rhodei ( https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SSIS33.1-75-7-1 ) but not for rhodep.

Thanks in advance.

asbraga-motorola, 2024-03-25:
Code published: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-T1SUS33.1-124-6-7

13riki, 2024-03-31:
> ID: <MotorolaMobilityLLC/kernel-msm/issue/596/issue_event/12241778298@
> github.com>

Deseo canselar esta subcriccion porfabor no deseo estar mas con esta
subscriccion